ELynx/ddf-350-modding #1: Enable multiplier MIX

dhilung, 2024-06-23:
Existing two MIX options are additive. I.e., `output(S) = M + S`. Can we change the second MIX such that `output(S) = M * S`.

The additive mix does not work well when you want to "scale" a channel output (e.g., throttle `CH2`) by a variable knob channel (e.g, `VR5`). My use case is exactly this. I'd like to quickly scale or attenuate the throttle channel directly using one of the variable knob as I switch the controller back and forth with my toddler :)

dhilung, 2024-06-24:
If anyone interested, a workaround is achievable with the current mod using following settings that allows your throttle to be scaled to `25%`, `50%`, `75%`, and `100%` by enabling and disabling `MIX1` and `MIX2` using `No1` and `No2` triggers.
\
[image]
\
`MIX1` doubles the current throttle while `MIX2` reduces the current throttle by half. Since mixtures are processed in parallel and not in series, 4 variations are possible instead of 3.

| MIX1 | MIX2 | Scale |
|--|--|--|
| 0 | 1 | 25% |
| 0 | 0 | 50% |
| 1 | 1 | 75% |
| 1 | 0 | 100% |

ELynx, 2024-06-27:
Hi! Apologies for late response, _somehow_ I missed the notification.

I was thinking about multiplication a lot myself. I often find that I want to have "direction" button, so I can quickly reverse a channel. Multiplying by -1 / 0 / 1 does that (and more).

My mind often rotate around 4ws, so for me it would be 3-pos button as multiplier. Center is zero, front only. Then -x% and +x% so I either scale rear steering with + or - sign. But I get your point to have "top speed" adjustment.

There is an issue with this, as far as I can think of. There needs to be "input" channel, in this case throttle, "scale" channel, such as VR-5 and VR-6 and "output" channel where value goes. But UI has only settings for two, most useful "input" and "scale".
I see a solution in writing output back to input. I see that "out of the box" this is _kind of_ supported, as you show in your workaround.

My solution, if I come to make one, would be: write "output" back to "input". It seems to be supported by remote logic in general.
I can make MIX1 "additive" so channel can be replicated, and MIX2 "multiplicative" so scale can be applied.

I will add this idea to my list, and to my "thought shelf". Hope workaround works for you in the meantime. *Maybe* I can hack into UI logic and add second round of "CHn" inputs, made as "x CHn" and have free selection between additive and multiplicative. No promises here, since mixing logic is quite messy.

TL;DR version :)
I had similar ideas, now that you ask about it and give your case, I will see what I can do.

82munk, 2024-07-04:
Hi. 
Hi hard is this to do? 
I need to do it & have -noob points😅 
I know you said you cant share your file maybe there's other laws too that you can't help with on your thread, so i started a Fork (plz tell to remove if an issue)
1] I dont know what program i need to flash with?
2] do i need to edit any files myself?

82munk, 2024-07-04:
[image]

The files bin files from dumbo are not recognised by PowerIso or 7zip or do i need another program?

ELynx, 2024-07-04:
@82munk , hi, I created a standalone issue to answer you, see my answer there

ELynx, 2024-07-15:
Looking into this; especially into UI parts of this.

ELynx, 2024-08-03:
Hey @dhilung 
Took me some time and some detours to learn, but I did multiplicative mixes, now released as Alpha
https://github.com/ELynx/ddf-350-modding/releases/tag/v0.0.4-alpha.3
It works pretty stably, I use it on my remote, but I wanted to know your opinion before making a release.
For flashung, make sure to apply patch to pristine original file, not already patched. I will update instructions to reflect this with release, so I give you a heads-up here.
I set up SW-7 to be my "kids mode" switch with 25 50 75 percent power. With progmix off it is 100 percent power.
You of cource can use fast toggle or keep it in menus for safety.
[image]
[image]
[image]

PR #7 (2024-08-09): Release 0.0.4
Closes #1 on basic level.
Closes #4 on basic level.

New features:
Allow "multiplicative" ProgMIX.
Add 4 NiMh main screen battery circle range.
closed 2024-08-09 by ELynx