ELynx/ddf-350-modding #2: ![20240704_130955.jpg](https://github.com/ELynx/ddf-350-modding/assets/174704962/972938f0-af62-4dbf-9a18-79aa33872bf5)

ELynx, 2024-07-04:
[image]

The files bin files from dumbo are not recognised by PowerIso or 7zip or do i need another program? 
Your patches look fine

What drivers do i need ddf350 & pc i get not recognised pop up

_Originally posted by @82munk in https://github.com/ELynx/ddf-350-modding/issues/1#issuecomment-2208857595_

82munk, 2024-07-04:
How do i find it?  Idk github

ELynx, 2024-07-04:
Wait man, I am writing :)

ELynx, 2024-07-04:
Since you are online now, can you tell me what you want to do with 7zip?

82munk, 2024-07-04:
Your mod & patch are fine, all the bin. files from dumbo are not recognised by PowerIso or 7zip, am i missing driver's?

ELynx, 2024-07-04:
OK, what have you tried in following these steps?
https://github.com/ELynx/ddf-350-modding?tab=readme-ov-file#how-to-install
At what step exactly you need 7zip?

I just tried 7zip on original .zip file and fresh download of 7zip, opened fine for me.
So, you take original .zip, extract somewhere. This is only step you need 7zip or any other .zip file.

Then you use LunarIPS, it is path on .bin.

You do not need to zip it back, or anything. Once .bin is patched, you just run .bat file near it.

For drivers: none that I know of. Take out any and all batteries, connect to USB, press CH3 and CH4, then power.

ELynx, 2024-07-07:
OK, I see no follow-up, hope it helped, closing.

closed 2024-07-07 by ELynx

ELynx, 2024-07-07:
Also did a video on how to do it
https://www.youtube.com/watch?v=8Tg5iURBdxE