ELynx/ddf-350-modding #4: feature request: nimh voltage definition

hicwic, 2024-07-26:
Hi,

first, congratulation for your work, i really like seeing project like this.

i have a feature request if you don't mind, and i really don't know the complexity of doing this.

Can you add an option in battery configuration to use NIMH. As 4 nimh are 1.2Vx4, the voltage displayed on main menu is never accurate.... 

thx again

ELynx, 2024-07-27:
Hi! Thanks!
I don't mind any suggestions, feel free to add them, and thanks for using Github for them, it is much cleaner here.
I will see if _adding_ is possible, because so far I usually replaced stuff. But this will be a good scope to see about adding.
As usual, no promises on speed or delivery :) but ya, noted!

ELynx, 2024-07-27:
@hicwic Hi, can you try this version?
https://github.com/ELynx/ddf-350-modding/releases/tag/v0.0.4-alpha.1
Logic is the same, apply patch, flash.
I used range of 3.6 to 5.6 volts. I tested it as much as I could but I do not have such batteries.
Also, while testing, make sure to apply patch to original firmware file, not previously patched. This step is absolutely necessary.

ELynx, 2024-07-28:
I will also see if I can adjust the range of battery alarm, it is now 4.5 volts lowest, but 13.0 volts highest. Maybe I can set it to 3.5 volts at the bottom at full release.

hicwic, 2024-07-28:
That was fast. Thank you to take my proposition in consideration.

Flashed with fully charged 4 nimh, eneloop brand. Work like a charm

[image]

For information, Voltage read from my multimeter is 5.7 with no load, and 5.55 with radio turned up

If you can lower the alarm to 3.6 it will be perfect.

Once more, thx for your work.

hicwic, 2024-07-28:
unfornutaly, control does not work properly anymore. There is a mix, between steering and throttle i can disable.

i disable it, but it reativate instantly when throttling
[image]

ELynx, 2024-07-28:
Got it.
Turn off "trigger priming" near place where you choose buttons to turn prigmix on and off.

hicwic, 2024-07-28:
nice, you'r right, this is OK now 👍 

thx again

ELynx, 2024-07-29:
I found a spot in the logic that operates the UI, so _in UI_ I can set up any value I want to. I chose 3.2 to 13.0 volts, only changing (lowering) the bottom value. BUT for some reason values below 4.1 are replaced by 4.5 on startup; What is worse, I could not find the spot where this is enforced, and _maybe_ it is in bootloader. And I am strictly against patching bootloader.
So, very likely for 0.0.4 release I will set alarm to 4.1 volts.

ELynx, 2024-08-01:
I tried once or twice more, it seems for now it will be 4.1 and above.
I made an issue memo for myself, so I will keep an eye on this.
I will make 0.0.4 alpha 2 release, you can use it until I release 0.0.4 proper.

ELynx, 2024-08-03:
@hicwic Hey, I see no responses for some time, I assume that everything is working.
I made an issue to allow lower voltage threshold #6 but in the meantime, can you with multimeter tell me as low as it would even start. I have a thought that below 4 volts it may not even work, but all of my referenses only go as low as 5, so I have no way of testing.

hicwic, 2024-08-03:
Yes sorry for not replying. 
I'm on holiday, not home, and I don't have the transmitter right now.
Can you wait maybe at the end of the month ?
Will try take it with me like middle of August...

ELynx, 2024-08-04:
> Yes sorry for not replying. I'm on holiday, not home, and I don't have the transmitter right now. Can you wait maybe at the end of the month ? Will try take it with me like middle of August...

Ya, sure, no rush. There is no "last bus" factor anywhere. If I make 0.0.4 release without this, I will make 0.0.5 release :) later.
I am on no schedule. I will also see if I can source variable voltage source somewhere to test lower limit.

closed 2024-08-09 by ELynx

hicwic, 2024-10-16:
hi, completly forgot to answer you.

i'm using your firmware with this nimh voltage mod since 2 month. It work like a charm

ELynx, 2024-10-19:
Oh, nice! I ported my changes to 1.1.3 as well :)

PR #7 (2024-08-09): Release 0.0.4
Closes #1 on basic level.
Closes #4 on basic level.

New features:
Allow "multiplicative" ProgMIX.
Add 4 NiMh main screen battery circle range.