ELynx/ddf-350-modding #5: 4WS rear drop out

jgillilan, 2024-07-26:
Hello, Great work! I installed the firmware and everything seems fine. When I select channel 3 for tight turning it turns as expected until I drive forwards, the rear wheels straighten and then don't work until I turn the mix off then it will temporarily work. Pretty sure I set the options exactly as in your video. The channel 4 crab walk seems to operate correctly.

jgillilan, 2024-07-26:
Looks like when I trigger channel 2 to drive it presses mix 1 for me? But only mix 1 it leaves 2 alone.

ELynx, 2024-07-27:
Hi,
Can you add pictures of your settings? It looks like some combo of settings. Check the "Trigger priming", I would say it should be off in your situation. Then also check buttons assignment.

jgillilan, 2024-07-27:
Bingo! Turned off the trigger priming and everything works great. Thanks
for your work on this!
I’d like to donate to the cause. What’s the best option for you?

ELynx, 2024-07-27:
Hey, glad it helped!
On the main page of this project, there are links to ko-fi, github sponsorship, check which one works for you.
Here they are also
https://ko-fi.com/eduardlynx
https://github.com/sponsors/ELynx
both should have one-time and repeated options.
If none of that works, we can work out paypal.

closed 2024-07-27 by ELynx

ELynx, 2024-08-04:
> Bingo! Turned off the trigger priming and everything works great. Thanks for your work on this! I’d like to donate to the cause. What’s the best option for you?
> […](#)

Hey, I have very strange readings on Ko-Fi, ones I don't understand. At one side, I see someone donated, at other side I cannot see who and when. If you did, is everything OK on your side? Sometimes paypal pauses these payments as "suspicious".