ELynx/ddf-350-modding #9: possible to add Model quick setting or bind to channel? ( don't know where to put a request )

Up50lut3, 2024-08-21:
Hiya!
Was wondering if it would be possible to add a Model switching quick setting or bind to channel? maybe channel 7 to switch between 3 different models or to a regular (like 3 4 8 9 OR 10) this would be a wonderful addition if possible.

either way, thank you for all the work you put into this already!

ELynx, 2024-08-21:
Hi!
I will see if this is possible :)
As usual no promises of results or speed, but it is good to keep backlog of items to do.

Up50lut3, 2024-08-22:
no worries at all. thank you for your time. if you'd like id be more than willing to spend some time on this also. i don't know much about firmware's, but i do do ECU/ecm tuning via hex editing and win OLS for address hunting usual 8 and 16 bit.( bosch me7x) doubt that's useful, here tho. 
The only reason i haven't tried changes i had made myself is due to i have no idea about check summing or its requirements on things of this nature for items as this.

Up50lut3, 2024-08-22:
also going to pull the U.I and make some changes there. i'll gadly give them to you once i finish a template up

ELynx, 2024-08-22:
> no worries at all. thank you for your time. if you'd like id be more than willing to spend some time on this also. i don't know much about firmware's, but i do do ECU/ecm tuning via hex editing and [...]

Do you need my Ghidra files? It is nothing extravagant, but at least I marked the areas where I change stuff. I do not have Ghidra server, I just zip the folder from time to time :) In general this is ARM LE and ghidra does pretty good job of it.

> also going to pull the U.I and make some changes there. i'll gadly give them to you once i finish a template up

NO WAY 😮 I could find some icons in there, but I never could figure RGB channels. I always ended up with some very strange red channels. But I gave it only a cursory look. I found that font colors are actually hex values, but nothing else. Maybe with some help I can bang together UI customizer.

ELynx, 2024-08-22:
@Up50lut3 I dropped you an invite to repo with Ghidra. About security - firmware has none of it :)
Literally not even checksums. It accepts whatever and does it's best to run it :)
If you have experience, and with Ghidra, it is quite straightforward to do once you have a clear goal.
If you are interested to work on this collaboratively, accept the invite, drop an issue in private repo, and we can share contacts there, such as TG or Discord.

Up50lut3, 2024-08-23:
> @Up50lut3 I dropped you an invite to repo with Ghidra. About security - firmware has none of it :) Literally not even checksums. It accepts whatever and does it's best to run it :) If you have [...]

hey man, thanks for the setup stuff and repo! I'm rebuilding my main PC tomorrow so once i have that setup, I'll start pouring over everything. i took a quick look and i appreciate what you've done and labeled( thank god!) 
Unfortunately I don't have any experience with Ghidra, but I'm a fairly quick learner.

Sure. id love to collaborate on this. Once i make some traction in  getting through the code, I'm going to make a second branch of your files and split that between images and functions; they will be shared as to not conflict with the original alphas and betas of your so I can highlight changes and not mix them, so it can stay in a structure of Your modifications- my/our modifications and then you can implement to yours/mains as you see fit. thanks again Looking forward to this!

ELynx, 2024-08-23:
> > @Up50lut3 I dropped you an invite to repo with Ghidra. About security - firmware has none of it :) Literally not even checksums. It accepts whatever and does it's best to run it :) If you have experience, and with Ghidra, it is quite straightforward to do once you have a clear goal. If you are interested to work on this collaboratively, accept the invite, drop an issue in private repo, and we can share contacts there, such as TG or Discord.
> 
> Hey man! Thanks for the setup stuff and repo! I'm rebuilding my main PC tomorrow so once i have that setup, I'll start pouring over everything. i took a quick look and i appreciate what you've done and labeled( thank god!) Unfortunately I don't have any experience with Ghidra, but I'm a fairly quick learner. picked up IDA-pro fairly quickly
> 
> Sure. id love to collaborate on this. Once i make some traction in getting through the code, I'm going to make a second branch of your files and split that between images and functions; they will be shared as to not conflict with the original alphas and betas of your so I can highlight changes and not mix them. It can stay in a structure of Your modifications- my/our modifications and then you can implement to yours/mains as you see fit. Thank you again for the opportunity to partake. Looking forward to this!

Super, works for me well. We can organise on how to apply changes. Never got to work with IDA-pro, because of license. IIRC there was an option to open Gidra files in IDA, as plugin. Right now latest changes are in 0.0.5a1.

ELynx, 2025-07-02:
Ran down through model change logic. It seems to be not offset-based, but rather replace-based.
In my understanding, whenever next model is selected, there is a routine that copies parameters from general memory to working addresses. So "quick" will be quite a painful process of recalling a long procedure and the forced refresh.
For now - out of scope.

closed 2025-07-02 by ELynx