ELynx/ddf-350-modding #20: Porting to 1.1.4 progress

ELynx, 2024-10-19:
* [ ] Wait for official 1.1.4 release

First of all, I promised to keep 1.1.4 preview private, and I keep my promises. Second, preview may change before release, and as such some of items here can become obsolete or require rework.

Definitely / By release

* [ ] EN translation
* [ ] DE translation
* [ ] 4NiMh battery display option

Very likely / As priority after release

* [ ] Toggle 4WS with single button
    * Will need to think on toggle cycle since now "Rear only" mode is a thing
    * This was and is a valid case, I liked how I can go through operation modes of 4WS crawler with single button
 
Definitely not

* Multiplicative mixing / "Kid mode"
    * All responses I got were about people using it as Dual rate or "Kid mode", no one mentioned to use it as multiplicative mixing for some specific setup
    * Even my case for using it is very 🤔 and I feel I cannot justify spending much time on it while I have more interesting and useful stuff on the plate
 * Toggle Progmix 1 and 2 with single button
    * All responses I got were about 4WS, toggling idea came from that direction as well
    * Again, I cannot justify spending much time while I have other things

ELynx, 2024-10-29:
Waiting for binary...

ELynx, 2024-11-10:
Public file received, let's go.

Lanfir3, 2024-11-17:
Hey there, do you know, is it possible to run this on a absima RC10P? i cant find firmware for this modell.
Should be the same because it is branded as DUMBO RC cooperation.

Lanfir3, 2024-11-19:
Hey Eduard,Thabk you for your reply. I contact absima but they didnt answer my question… so the support is poor. I‘ll wait until february, maybe i get an answer. So thank you for your support. BrMarkusGesendet von Outlook für iOS Von: Eduard ***@***.***>Gesendet: Dienstag, November 19, 2024 1:51 PMAn: ELynx/ddf-350-modding ***@***.***>Cc: Lanfir3 ***@***.***>; Comment ***@***.***>Betreff: Re: [ELynx/ddf-350-modding] Porting to 1.1.5 progress (Issue #20) Hey there, do you know, is it possible to run this on a absima RC10P? i cant find firmware for this modell. Should be the same because it is branded as DUMBO RC cooperation.Hmmmm, from one site it seems possible, since visually this is same transmitter, just tad more stickers on the bottom. On the other side, no idea which "tricks" they added inside, if any. I personally do not have it, so I cannot try it.From https://absima-my.sharepoint.com/personal/office_absima_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Foffice%5Fabsima%5Fcom%2FDocuments%2FAnleitungen%2DDatenbl%C3%A4tter%2Faktuelle%20Anleitungen%2F2000013%2Dcr10p%2Dradio%2Dmanual%2Dde%2Den%2Dfr%2Epdf&parent=%2Fpersonal%2Foffice%5Fabsima%5Fcom%2FDocuments%2FAnleitungen%2DDatenbl%C3%A4tter%2Faktuelle%20Anleitungen&ga=1 I see it is 1.0.9 without without any special items. Someone daring could try update path from 1.0.9 to 1.1.5 with all the boot updates and so on (remember to take out the battery) and tell the tale.Or, they have contact data, Telefon: +49 911 650841 30 email: ***@***.*** , ask them their opinion :)

Lanfir3, 2024-11-19:
Yeah but i cant race outside until next year, its freezing cold outside. I managed to build a adruino board (with tutorial) to connect my remote via usb and race the vrc simulator. So i‘m quite happy for this solution. I would try to update my firmware but i‘m afraid for bricking it. Dumborc is also not responding on my ticket :(So i‘m waiting. I would be happy if you get any news from dumbo and let me know :)Gesendet von Outlook für iOS Von: Eduard ***@***.***>Gesendet: Dienstag, November 19, 2024 7:49 PMAn: ELynx/ddf-350-modding ***@***.***>Cc: Lanfir3 ***@***.***>; Comment ***@***.***>Betreff: Re: [ELynx/ddf-350-modding] Porting to 1.1.5 progress (Issue #20) Hey Eduard,Thabk you for your reply. I contact absima but they didnt answer my question… so the support is poor. I‘ll wait until february, maybe i get an answer. So thank you for your support.
Out of sheer curiosity, why February? It is quite a while to wait :)I have not seen that remote before, so I assume it is new. I think someone will sooner or later try to do it.From my side, I will try to ask DumboRC contact(s) I have, maybe that will resolve faster. My idea is, 1.1.5 is a big thing, mods or not, and DumboRC would like that to be used by more people.

ELynx, 2024-12-02:
> Yeah but i cant race outside until next year, its freezing cold outside. I managed to build a adruino board (with tutorial) to connect my remote via usb and race the vrc simulator. So i‘m quite [...]

Hey, I got an answer back from Dumborc on their BiliBili

> We customized the transmitter for them, which can be brushed with the 1.1.5 firmware of our original factory

I would consider that 85% affirmative :)

So, if you feel brave, you can try doing it.
Remember, 100% take out the batteries, even if it worked before with batteries, it will not.
Take batteries out.
Then, find the version you have now in "Transmitter" -> "About" section.
Below 1.0.9 -> boot 1.0.9, boot 1.1.5, firmware 1.1.5
1.0.9 and above -> boot 1.1.4, firmware 1.1.5

In doubt, ask me :)

Lanfir3, 2024-12-04:
Hey Eduard,Thank you for this feedback!I‘ll try that in the christmas time. Now i‘m very busy and have no time for my rc. Much of work and to do‘s…But i‘ll give you an update when im done 👍Gesendet von Outlook für iOS Von: Eduard ***@***.***>Gesendet: Montag, Dezember 2, 2024 10:21 PMAn: ELynx/ddf-350-modding ***@***.***>Cc: Lanfir3 ***@***.***>; Comment ***@***.***>Betreff: Re: [ELynx/ddf-350-modding] Porting to 1.1.5 progress (Issue #20) Yeah but i cant race outside until next year, its freezing cold outside. I managed to build a adruino board (with tutorial) to connect my remote via usb and race the vrc simulator. So i‘m quite happy for this solution. I would try to update my firmware but i‘m afraid for bricking it. Dumborc is also not responding on my ticket :(So i‘m waiting. I would be happy if you get any news from dumbo and let me know :)Hey, I got an answer back from Dumborc on their BiliBiliWe customized the transmitter for them, which can be brushed with the 1.1.5 firmware of our original factoryI would consider that 85% affirmative :)So, if you feel brave, you can try doing it.
Remember, 100% take out the batteries, even if it worked before with batteries, it will not.
Take batteries out.
Then, find the version you have now in "Transmitter" -> "About" section.
Below 1.0.9 -> boot 1.0.9, boot 1.1.5, firmware 1.1.5
1.0.9 and above -> boot 1.1.4, firmware 1.1.5In doubt, ask me :)

Lanfir3, 2025-01-02:
Hey Eduard,I‘ll tell you that i upgraded the Firmware to the newest Dumborc.Now i have dualrate also, which was not supportet in early versions. I took the risk because of that 🙂I‘am now racing a Atomic Amz 4wd and its amazing. Also i wish you a happy new year and thank you for support 😌Br,Markus
    
    Gesendet von Outlook für iOS

ELynx, 2025-01-02:
> Hey Eduard,I‘ll tell you that i upgraded the Firmware to the newest Dumborc.Now i have dualrate also, which was not supportet in early versions. I took the risk because of that 🙂I‘am now racing a Atomic Amz 4wd and its amazing. Also i wish you a happy new year and thank you for support 😌Br,Markus

Hey! Congratulations!
I will add this to my list of information. I expect more people will be interested in this, as Absima propagates around.

Best of wishes to you too, and good driving :)

closed 2025-06-23 by ELynx