ELynx/ddf-350-modding #27: Implement an anti-lock braking system function

bryankroh, 2025-07-15:
Many controllers offer an anti-lock braking system function (ABS), such that when the brake is applied it doesn't fully lock the wheel causing a skid, but rather transmits quick intermittent braking commands so that the car "pulses" to a stop without locking up, like a modern day automobile does.

ELynx, 2025-07-16:
https://github.com/user-attachments/assets/1309c88b-2ba3-4500-8906-2d8ef55c773f

Very good explanation video!

I see my solution there as follows:

1. I add a progmix type "ABS".
2. As usual Lead and Follow channels are unlocked. If someone wants to do a one-side-jittery steering they are free to do so :) but main goal is Lead CH10 Follow CH2.
3. Using new Servo Sweep, CH10 is set to oscillate. Kumpel Frank has a [video](https://www.bilibili.com/video/BV1dW7izWEqg/) in Chinese. I don't understand what he says, but I can follow the pictures good enough.
4. In short, CH10 oscillates between two values, e.g. 0 and 30%, with period of about 0.1 second, quite alike to what I saw in the video. It can go slower, and I will see if I can make it go faster.
5. A new setting "ABS" on one of Progmixes adds value of CH10 to CH2, when throttle is below e.g. "negative" mix value.

So,
* ✅ Brake return is covered by Servo Sweep settings, in this case it even goes a bit deeper with both "high" and "low" of extra signals unlocked and being positive and negative
* 🔴 Delay - no. Servo Sweep has no such thing, and time logic on the radio loop there is _something_. Maybe some time in the future, but at first iteration no.
* ✅ Cycle Length. Ya sure, Servo Sweep has it, can be set from VR or toggled by key. Currently at 0.1 seconds, will see if can go lower.
* ✅ Trigger Point - very much a must. Without it, whole thing is useless. Will see if throttle above 100 or brake below 100 is possible, but in general for -100/100 should be settable.
* 🔴 Duty cycle - no. Again, Servo Sweep does not support that.
* ❓ Steering mix - not at first. While not outright impossible, I would rather focus on getting it out as usual thing.

Side effects, at least for 1st iteration: constantly jerking CH10 arrow, Servo Sweep engage is a mess IMO. If this gains traction, will see if special engage can be used instead of key.

Alternatively, I will see if I can make Servo Sweep into standalone ABS logic and just sum it up using usual progmix. In general, cool idea.

Now, personally, is this idea a suggestion or an actually feature that you would use? I myself have no basher and no ESC that supports this, so I will not be able to check actual usefulness beside some sacrificial servo.

ELynx, 2025-07-18:
OK, then I will not rush to do this, since I usually prioritize things with practical applications. But not to close this, since it seems to be rather doable. If you _do_ get the model that supports this, ping me here again :)